rdc-autonomous-mobile-robot-project/tsukuba2023_ToDo #12: 信号認識

yuzukiimai, 2023-04-26:


yuzukiimai, 2023-09-12:
- [ ] yolov5の環境構築
- [ ] アノテーション
- [ ] 信号の認識（青・赤）

haruyama8940, 2023-10-31:
https://github.com/rdc-autonomous-mobile-robot-project/YOLO-v5-ROS/tree/traffic
このブランチに処理を追加しました．
・traffic_light(Trgiger)のサービスを受けると，detectionを開始します．
・信号認識の結果が，赤から青に変わるとnext_wpをwpに投げ，detectionを終了します

masakifujiwara1, 2023-10-31:
@haruyama8940 
https://github.com/rdc-autonomous-mobile-robot-project/tsukuba2023_ToDo/issues/16#issuecomment-1787020346
ここに記載したように，trafic_light用のブランチ（[tsukuba2023-trafic-light](https://github.com/masakifujiwara1/waypoint_manager2/tree/tsukuba2023-trafic-light)）を作成しました．

以下の機能を追加しています．

- trafic waypointの追加（青）
機能としては，stop waypointにサービスを投げる機能を追加しただけです．

- ロボットが停止後，traffic_light（Trigger）のサービスを投げる．
- /next_wpが投げられると，次のwaypointに向けて，navigationを再開する．
万が一，認識が上手く行かなかった際の手動トリガーにも対応