rdc-autonomous-mobile-robot-project/tsukuba2023_ToDo #43: 11/17-19_つくば5回目+本番

masakifujiwara1, 2023-11-11:


masakifujiwara1, 2023-11-11:
waypointに関して
- #42 
- #41 
を行った．

masakifujiwara1, 2023-11-11:
横断歩道における自己位置推定の破綻に関して，以下のissueで議論した．
- #40

Ryusei-Baba, 2023-11-17:
11/17議論

- ypspurがdisconnectしてしまう
→　dmsgsコマンドによりUSBの接続（モータドライバとJetson）を確認した結果，4回ほど接続が切れていた．
→　接続が切れていたことから，ソフトウェアではなくハードウェアに問題があるのではないかと考えた．
→　ケーブルの変更により改善を試みる．また，udevadmによりその様子を監視する．

- 車道を走るところで自己位置がずれてしまう
→　pointcloud_to_laserscanのパラメータが地図を作成したときとは違うため（変更したため）自己位置がずれた．
→　パラメータをもとに戻したら自己位置推定が改善した様子が見られた．
→　このことから地図作成時とナビゲーション時のパラメータはあまり変えないほうが良さそう．

masakifujiwara1, 2023-11-23:
1119

結果として，スタート直後（13m）で失敗となった．
以下に本走行時のrviz画面を見ると，前方のチームの随行員がコストに反映され，ロボットが停止した際に自己位置が誘拐されている．
https://drive.google.com/file/d/1A6VaacHUzDp3QAel9AHilxe8lgcKg2tr/view?usp=drive_link

masakifujiwara1, 2023-11-23:
2023 つくばチャレンジ総評

良かったところ
- ROS2で自己位置推定を行い，自律移動するノウハウが得られた

改善点
- タスク表での情報共有が不足していた
- IMUの問題を認識していながら，その問題に取り組めていなかった
- 選択と集中を行い，早い段階で最低限の安定した自律移動ができるようにすべきだった

masakifujiwara1, 2023-11-23:
1118

- カーソルを大きくする設定をしたところ，画面を正常に操作することができなくなった